vshymanskyy/ViperIDE #1: Error on connecting Pyboard, no file display

rkompass, 2024-06-13:
Hello @vshymanskyy,

thanks for suggesting to try out WebSocket/WebREPL. I'll try it tonight or tomorrow.

When connecting Pyboard (through dev/ttyACM0 here) there were a few errors like:

```
Error: Cannot read properties of undefined (reading 'split')
Error reading board info
```

```
Error: Cannot exec command: Traceback (most recent call last): File "", line 11, in File "", line 10, in walk File "", line 3, in walk OSError: [Errno 22] EINVAL
Execution failed .
```
And consequently the file manager on the left side does not show any files.

Did you try ViperIDE with a pyboard?

I tend to assume this is a problem of the pyboard MP implementation (I used the latest preview, and 1.17 (old version still there before)), as also e.g.
`mpflash list` which is supposed to report the board fails with:
`Error: Failed to get mcu_info for /dev/ttyACM2` ?

I'll report at mpflash site too.

Again, no need to hurry, btw., good work takes it's time.

vshymanskyy, 2024-06-13:
No, unfortunately I have lost all my pyboards 2 years ago, so unable to test with them.
I will definitely try to find a way to make the original pyboards working :)

vshymanskyy, 2024-06-14:
@rkompass please try again, I've done some major changes to the core functions

rkompass, 2024-06-14:
The improvement I see is: No more a Micropython banner at every run. Which is good.

Do I recall correctly that you use `mpremote` under the hood? I noticed an issue there and [reported](https://github.com/micropython/micropython/issues/15282) it.

Yet no FS display.

vshymanskyy, 2024-06-14:
No, it doesn't use mpremote under the hood :)

rkompass, 2024-06-15:
I tried the `webrepl` from micropython-lib as well as your `web_repl` (this repo, nicer) and both serve a html screen as expected.
Yours asks for the password too.
At connecting however it says `Cannot connect at at the MP promt `. At the promt of the w600 there is this error message (with both variants of web_repl/webrepl) :
 ```
WebREPL connection from: ('192.168.178.46', 36136)
dupterm: EOF received, deactivating
```
This is an [older problem](https://github.com/micropython/webrepl/pull/66), but I don't know how to solve it.

vshymanskyy, 2024-06-15:
Oh! You were using the W600!!!
I just verified it with this chip, everything seems to when connected via USB.
WebREPL (via the `viper-repl` package) fails with the same symptoms as yours. This is a W600-specific bug, as it works on ESP32 and Pi Pico W.

vshymanskyy, 2024-06-15:
How about the Pyboard?

rkompass, 2024-06-15:
> This is a W600-specific bug, as it works on ESP32 and Pi Pico W..

Good to know. I will report it to Robert. Perhaps he/we can find a solution.

vshymanskyy, 2024-06-19:
I have debugged it using `MPY v1.23.0 (2024-06-02)` with this board: https://micropython.org/download/ADAFRUIT_F405_EXPRESS/
This should be very similar to Pyboard v1.0 and v1.1 (it uses the same MCU).

- The file tree - **fixed**
- File open/save - works
- Run script - works
- Package installation - **fixed**

Could you please re-check it with `ViperIDE v0.3.1`?

rkompass, 2024-06-20:
I tested with a Blackpill board and it worked. My pyboard fails, unfortunately, still. (No file display). But note, that there was an error anyway (probably a frimware problem). I will try later with a different firmware.

vshymanskyy, 2024-06-20:
Looks like some of your files on the board have invalid unicode in their names. Try erasing your fs

vshymanskyy, 2024-06-20:
@rkompass did it help?

rkompass, 2024-06-20:
It helps. After a [factory reset](https://docs.micropython.org/en/latest/pyboard/tutorial/reset.html) of the filesystem viper ide is able to display the files. It also runs them. So everything seems fine.

Now the w600 has a working webrepl again (regression solved). Btw. Files structure are displayed but not file contents....

vshymanskyy, 2024-06-20:
Works on my end. Is it pyboard or w600, or both?
What exactly happens?

rkompass, 2024-06-20:
Pyboard works, as you already announced.

The Winner w600 has now a working webrepl again, and it works partly with viper ide.
Files structure are displayed but not file contents....

vshymanskyy, 2024-06-20:
I see, so this only affects webrepl on w600. I can check it on my end. Do i need to rebuild the w600 firmware?

rkompass, 2024-06-20:
Take that from robert-hhs [shared stuff](https://github.com/robert-hh/Shared-Stuff/tree/master/w600_firmware).

rkompass, 2024-06-20:
This issue may be closed now.

commit 84968c12 (2024-06-19): Fix compatibility with pyboards. Fixes #1
closed 2024-06-20 by vshymanskyy