vshymanskyy/ViperIDE #4: No micropython specific syntax highlights

Josverl, 2024-06-19:
There is no awareness of what non standard modules exist in micropython, and it assumes the full cpython stdlib is available

vshymanskyy, 2024-06-19:
How does it "assume" anything about cpython stdlib?

vshymanskyy, 2024-06-19:
Also, please clarify what is wrong with MPY syntax.

Josverl, 2024-06-19:
the comemirrors syntax highlighter is using is based on CCython, and there are quite a few subtle and more significant differences in what is / is not supported in Micropython. https://docs.micropython.org/en/latest/genrst/index.html

This is quite hard to get right even in full fledged IDEs such as VSCode, Pycharm and also in Thonny.
but all have some level of support for type stubs , and that is where the micropython-stubs may be able to help.
as another sample of some of the differences you could look at the [validation tests for the stubs](https://github.com/Josverl/micropython-stubs/tree/main/tests/quality_tests) 
But if you want to keep things quick, clean and simple, there are no big problems waiting, perhaps just some end-user confusion.

It may be useful to add common Micropython keywords such as non standard modules and common GPIO Pin names to make life easier for your users.
ref: https://codemirror.net/try/?example=Custom%20completions

So just a reach out if you are  interested in integrating such functionality.

closed 2024-06-19 by Josverl

vshymanskyy, 2024-07-25:
Ruff is now used for code analysis in ViperIDE.
Custom Completions and keywords may be added later