vshymanskyy/ViperIDE #7: Timeout with Big files tree

picpic020960, 2024-06-20:
[re]Bonjour ,
On m'y watch V2 , Big files tree and 'timeout' message ; then the listdir on the REPL console.
So the file explorer IS empty.
I have old version of micropython (1.14) but the other fonctions are OK : run script , packages , soft/hard reset.

One trick ?
Thank's for help
NB : good job

vshymanskyy, 2024-06-20:
This is interesting... How many files do you have on your device?

vshymanskyy, 2024-06-20:
There's no particular interest in supporting outdated MPY versions. Why do you use v1.14? It's 3.5 years old.
Can you try a recent version?

vshymanskyy, 2024-06-20:
Please try again with 0.3.1 build 2024-06-20 22:32:12

Pedrocasf, 2024-06-21:
I can confirm that the issue still happens with the latest micropython preview v1.24.0-preview.47.g88513d122 (2024-06-19)  when using BLE on my ESP32-S3 on both Linux and Android.

vshymanskyy, 2024-06-21:
On the screenshot above, you're using a USB connection. Does it work?

picpic020960, 2024-06-21:
> Il n'y a aucun intérêt particulier à prendre en charge les versions MPY obsolètes. Pourquoi utilisez-vous la v1.14 ? Il a 3,5 ans. Pouvez-vous essayer une version récente ?

V1.14 with lvgl : not easy to build

rkompass, 2024-06-21:
Hello Volodimir,

I have this same error (Timeout reached before finding the ending sequence) when connecting the w600 through the web_repl.
The hierarchy of the few files on the board at first is displayed correctly. Terminal works.
But: When I double-click a file to open in the editor, (or create a new file, after entering it's name) then this error occurrs.
The terminal then also stops echoing my keystrokes.
At the w600 primary terminal (screen /dev/ttyUSB0 115200)  then I see (if file doubleclick)
```
raw REPL; CTRL-B to exit
>
```
is displayed without accepting any input (tried also the Ctrl-B: no reaction).

vshymanskyy, 2024-06-21:
@rkompass are you sure your issue is related?

@picpic020960 do you have an issue with BLE or USB connection, it is unclear.
Also, did you try with the latest version?

rkompass, 2024-06-21:
I can open a new issue for w600.
Thought it might be related as the error message (in the red box, right above) is the same.
I'm not sure, though.

picpic020960, 2024-06-21:
Better !
File explorer OK AT begin.
Load file OK but save After update KO : file size = 0
New file : save KO also After re-connect.

Improvement : zoom and proportionnal font : it's better for yes on Android tablet

vshymanskyy, 2024-06-21:
The initial issue was resolved, so closing this. Please create new tickets for each problem