vshymanskyy/ViperIDE #8: update file

picpic020960, 2024-06-21:
**Describe the bug**
A clear and concise description of what the bug is.

**Setup**
 - MicroPython board: [e.g. ESP32, RPi Pico W]
 - MicroPython version: [e.g. 1.23.0]
 - OS version: [e.g. Windows, Android 13]
 - Browser: [e.g. chrome, safari]
 - Connection type: [e.g. USB, BLE, WebREPL]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. ....
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Actual Screenshots / Logs**
If applicable, add screenshots or logs to help explain your problem.

**Additional context**
Add any other context about the problem here.

vshymanskyy, 2024-06-21:
Please check with Viper IDE v0.3.2

vshymanskyy, 2024-06-21:
Yes this was a small regression in the recent change. Should be fixed now

vshymanskyy, 2024-06-21:
@picpic020960 can be closed?

vshymanskyy, 2024-06-22:
The project is awaiting your github star 😂

vshymanskyy, 2024-06-23:
@picpic020960 close?

closed 2024-06-24 by vshymanskyy