vshymanskyy/ViperIDE #11: File operations don't work with Chromium (Linux).

rkompass, 2024-06-25:
Hello,

I have the error: `Timeout reached before finding the ending sequence` when connecting with the Wemos w600 Pico (robert-hh s port) through the web_repl.
The hierarchy of the few files on the board at first is displayed correctly. Terminal works.
But: When I double-click a file to open in the editor, (or create a new file, after entering it's name) then this error occurrs.
The terminal then also stops echoing my keystrokes.
At the w600 primary terminal (screen /dev/ttyUSB0 115200) then I see (if file doubleclick)

raw REPL; CTRL-B to exit
>
is displayed without accepting any input (tried also the Ctrl-B: no reaction).

This problem only occurs on my Chromium browser (with Manjaro Linux).

On Firefox the file operations work o.k., as previously tested by Robert.

So perhaps a solution might be to complement the _Tested with_ - list by a list of browsers tested to work.

vshymanskyy, 2024-06-25:
Sometimes it works, sometimes doesn't. WebREPL is not stable on W600 in general, but it works on ESP8266, ESP32 and Pi Pico W.

rkompass, 2024-06-25:
Oh, on Firefox it was very stable today.I could do everything. So w600 port and your latest viper version seem fine. You can close this. Just wanted to tell you that there is heavy dependence on the browser variant and that might be emphasized in the intro/docs.--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

vshymanskyy, 2024-06-25:
@robert-hh this is an interesting observation

robert-hh, 2024-06-25:
I tested that with Google Chrome and it worked.

robert-hh, 2024-06-25:
I installed Brave on my Debian system, and the Viper IDE works fine with a W600. No problem with file operations.
Brave: [Version 1.67.119 Chromium: 126.0.6478.114 (Offizieller Build) (64-Bit)](https://brave.com/latest/)

vshymanskyy, 2024-06-25:
Try doing this:
- check you're using the latest w600 firmware build
- ensure only minimal WebREPL code is running, no custom application
- perform power reset
- try connecting cleanly

Usually it works after this but it fails if i reconnect to the device, maybe after several times (on my side)

rkompass, 2024-06-26:
Of course I did that. web_repl is started in main.py. After re-plugging the w600 viper-ide is there.

All (including file-) operations work with **Firefox** (Version 127.0 64-bit for Manjaro Linux) and **Brave** ([Version 1.67.119 Chromium: 126.0.6478.114 (Offizieller Build) (64-Bit)](https://brave.com/latest/)), from the WebRepl. 

Only terminal and first directory tree display with pure **Chromium** (Version 126.0.6478.114 (Offizieller Build) Arch Linux (64-Bit)), and **Opera**.

At an attempt to do file operations (after clicking a file for display, for example) the above error: 
`Timeout reached before finding the ending sequence` is diplayed in the upper right corner.

Some strange configuration detail with the browsers, I suppose. Probably not worth investigating too much.
No problem with w600 firmware anymore, as Robert stated already :-)

My suggestion again is to only document that, so that users aren't thinking their board or firmware might have a problem.

vshymanskyy, 2024-06-26:
Got it. Let's leave the ticket open for now

vshymanskyy, 2024-06-26:
Interesting! I'll try it later, probably Opera.

jeffstearns, 2025-04-05:
I also see the `error: Timeout reached before finding the ending sequence`.  My ESP32C3 always fails with this error.  _It doesn't matter what browser I use_: Chrome, Firefox, or Safari.  All three browsers suffer from this error.

Despite dozens of reboots and reinstalls, it always fails the same way:

1. The browser connects to the ESP32C3,
2. the browser loads the basic ViperIDE page,
3. the browser then posts the error message.

It looks like this happens while it's trying to load ViperIDE's "File Manager" pane.  That shouldn't be difficult -- there are only 7 files on the ESP32C3: boot.py, main.py, and the 5 library modules installed by viper-tools.  There's nothing else on the ESP32C3, and no other code running.  It's just a stock machine with the basic web_repl setup code copied directly from the documentation page.

The USB connection works like a champ every time.  Never a hitch.  But web_repl is the opposite: it has never worked.

As a test, I installed the stock webrepl from micropython-lib.  It works, but, ugh, it's primitive and missing so much of what ViperIDE offers.  So I can rule out networking and a broken ESP32C3.  The failure is specific to the ViperIDE version of web_repl.

For the record, here's what the Chrome console reports when the error occurs:

> (index):2 Error: Timeout reached before finding the ending sequence
>     at Vx.readUntil ((index):4:124229)
>     at async Yx.interruptProgram ((index):4:131844)
>     at async Yx.enterRawRepl ((index):4:132125)
>     at async Yx.begin ((index):4:131632)
>     at async I$ ((index):10:73253)
>     at async (index):10:90650 'Error: Timeout reached before finding the ending sequence
>    at Vx.readUntil (http://10.5.4.95:8266/:4:124229)
>    at async Yx.interruptProgram (http://10.5.4.95:8266/:4:131844)
>    at async Yx.enterRawRepl (http://10.5.4.95:8266/:4:132125)
>    at async Yx.begin (http://10.5.4.95:8266/:4:131632)
>    at async I$ (http://10.5.4.95:8266/:10:73253)
 >   at async http://10.5.4.95:8266/:10:90650'
> O_	@	(index):2
> interruptProgram	@	(index):4
> await in interruptProgram		
> enterRawRepl	@	(index):4
> await in enterRawRepl		
> begin	@	(index):4
> I$	@	(index):10
> await in I$		
> (anonymous)	@	(index):10
> await in (anonymous)		
> (anonymous)	@	(index):10
> (anonymous)	@	(index):10