vshymanskyy/ViperIDE #13: Secure WebSocket Relay connection working with W600 flawlessly. Minor issue.

rkompass, 2024-06-29:
Hello Volodymyr,

A first feedback, as you invited/requested:

I tested the new WebSocket Relay connection with the latest w600 firmware (and Brave) and it works flawlessly.
Of course there is a little bit less memory available now, but we have enough of it :-)
Also the repl is a bit slower, which makes sense.
I like the flicker of the webrepl symbol while transferring (saving, e.g.) files.
Also it's nice to have the _run_ triangle getting red while executing a script!

A minor issue, which was probably already there at the web_repl:
When changing a script, after saving, there is still the old size of it displayed in the directory view.
I have to switch to package manager display and back to directory tree display to get the updated size, which is annoying.

rkompass, 2024-06-29:
More feedback:
Positive:
With the wss_repl all my browser seem to work (also Opera and Chromium on Manjaro Linux).

Another potential issue:
A command like in the docs:
`wss_repl.start(uid=070R-19C3-4069) ` gives
`SyntaxError: invalid syntax for integer with base 10`

It is not clear to me, how to specify a fixed uid.

vshymanskyy, 2024-06-29:
Thanks for the feedback. Yes the file size is not updated currently. It's not easy to fix though...

vshymanskyy, 2024-06-29:
We can free some memory by moving the websocket client implementation to native code.

vshymanskyy, 2024-06-29:
With the recent changes, you'll need to reinstall the `viper-tools`. There are some breaking changes

vshymanskyy, 2024-07-02:
Should be fixed now

commit 34875844 (2024-07-02): Fix #13
commit 5fa41162 (2024-10-15): Use new package manager
Closes #13
closed 2024-07-02 by vshymanskyy