vshymanskyy/ViperIDE #17: version log

littlegreenbean33, 2024-07-29:
Hi,

I was just prompted with new 1.49 version but I see no branch activity in github. ?¿? 

Is there a  version log with improvement from the different versions ?

Thanks. Great work. I have been using it daily for the last 3 weeks and it is a significant efficiency imporvement.

vshymanskyy, 2024-07-29:
The version log currently not being maintained, as there are a lot of small changes and it's not well-structured. You can see the commit log though

vshymanskyy, 2024-07-30:
> I have been using it daily for the last 3 weeks and it is a significant efficiency improvement.

Could you please elaborate, what workflow have you been using and what's the improvement?

littlegreenbean33, 2024-07-30:
Well, instead of editing on my laptop,  downloading, executing.....reset... in my prototyping cycle, I just develop directly on the board using ViperIDE. Out of a click I can soft-reset.  I have a small script that via mpremote downloads my assets back to my computer for version control and snapshots. Nothing extremely sophisticated.

brainstorm, 2024-08-03:
> Well, instead of editing on my laptop, downloading, executing.....reset... in my prototyping cycle, I just develop directly on the board using ViperIDE. Out of a click I can soft-reset. I have a [...]

I second that analysis, ViperIDE is great and I've been recommending it to a few friends! 

I was about to write that `mpremote` script myself, can you share yours @littlegreenbean33? 

OTOH, I think that a direct integration with [github repository contents API](https://docs.github.com/en/rest/repos/contents?apiVersion=2022-11-28) could be a killer feature to avoid losing recent "local" board/host changes... context: I introduced a change on my board that locked me out completely, even `mpremote cp` didn't work. I had to resort to `esptool.py read_flash ...` to carve out the flash `.py` files with [radare2](https://github.com/radareorg/radare2/) to recover my latest changes from the flash dump.

@vshymanskyy If there was an integration with Github repos API, there would be no need for local filesystem/sync and changes could potentially be staged on save and committed whenever?

vshymanskyy, 2024-08-03:
The idea is cool 😎 
But, i'm wondering how the sync could work. In instance, i have a GitHub repo with some files. Now, i'm connecting a device that may have a slightly (or completely) different file system. Which one is correct? Is this device even associated with the repository?
Same applies to regular host filesystem access, if/when it gets implemented.

brainstorm, 2024-08-04:
Moving the GitHub repos to a separate issue (don't want to steal this thread): https://github.com/vshymanskyy/ViperIDE/issues/19

closed 2024-08-05 by vshymanskyy