vshymanskyy/ViperIDE #18: LAN webrepl timeout ESP32

ThinkTransit, 2024-07-30:
**Describe the bug**
When connecting webrepl on local LAN, viper IDE loads but timeout error is display and connection lost

**Setup**
 - MicroPython board: ESP32 Feather S2
 - MicroPython version: v1.24.0-preview.31.g3c8089d1b
 - OS version: Ubuntu
 - Browser: chrome
 - Connection type: WebREPL

**To Reproduce**
Steps to reproduce the behavior:
1. Follow instructions for installing webrepl, install viper tools, use sample main.py etc...
2. Browse to lan address, http://192.168.17.106:8266/
3. Wait for ViperIDE to load

**Expected behavior**
ViperIDE should load and display repl and all files on device.

**Actual Screenshots / Logs**
ViperIDE is loaded but no files/repl displayed
[image]

```
Error
Error: Timeout reached before finding the ending sequence
```

**Additional context**
USB repl output
```
>>> 
MPY: soft reboot
WebREPL server started on http://192.168.17.106:8266/
MicroPython v1.24.0-preview.31.g3c8089d1b on 2024-06-06; FeatherS2 with ESP32-S2
Type "help()" for more information.
>>> 
WebREPL connection from: ('192.168.17.250', 54284)
```

vshymanskyy, 2024-07-30:
Did you try the latest stable MicroPython build?

DeeJay, 2024-07-31:
_{If this is seen as a 'highjack' then please let me know and I will open a separate dedicated Issue.}_

I have the 'same' issue, but with a different target microcontroller and different browser client.

I am using a Raspberry Pi Pico W, running the 'stable' 1.23 release of MicroPython -
```
MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
```

My client browser is Microsoft Edge running on a Windows10 desktop.

The 'steps to reproduce' and failure mode are as previously described.

I have modified viper-tools/web_repl.py to set the Debug flag and some debugging print statements.

The following is logged to the Terminal window -
```
WebREPL server started on http://192.168.1.206:8266/
webrepl should be ready
MAIN terminating
MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
>>>  in function ACCEPT_CONN
b'GET / HTTP/1.1\r\n'Host: 192.168.1.206:8266
(b'Host', b'192.168.1.206:8266')
Connection: keep-alive
(b'Connection', b'keep-alive')
Upgrade-Insecure-Requests: 1
(b'Upgrade-Insecure-Requests', b'1')
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/127.0.0.0 Safari/537.36 Edg/127.0.0.0
(b'User-Agent', b'Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/127.0.0.0 Safari/537.36 Edg/127.0.0.0')
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.7
(b'Accept', b'text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.7')
Accept-Encoding: gzip, deflate
(b'Accept-Encoding', b'gzip, deflate')
Accept-Language: en-GB,en;q=0.9,en-US;q=0.8
(b'Accept-Language', b'en-GB,en;q=0.9,en-US;q=0.8')
Cookie: ajs_anonymous_id=63d76b69-c268-4b63-aeeb-704fd15e24bc
(b'Cookie', b'ajs_anonymous_id=63d76b69-c268-4b63-aeeb-704fd15e24bc')
 in function ACCEPT_CONN
b'GET /favicon.ico HTTP/1.1\r\n'Host: 192.168.1.206:8266
(b'Host', b'192.168.1.206:8266')
Connection: keep-alive
(b'Connection', b'keep-alive')
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/127.0.0.0 Safari/537.36 Edg/127.0.0.0
(b'User-Agent', b'Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/127.0.0.0 Safari/537.36 Edg/127.0.0.0')
Accept: image/avif,image/webp,image/apng,image/svg+xml,image/*,*/*;q=0.8
(b'Accept', b'image/avif,image/webp,image/apng,image/svg+xml,image/*,*/*;q=0.8')
Referer: http://192.168.1.206:8266/
(b'Referer', b'http://192.168.1.206:8266/')
Accept-Encoding: gzip, deflate
(b'Accept-Encoding', b'gzip, deflate')
Accept-Language: en-GB,en;q=0.9,en-US;q=0.8
(b'Accept-Language', b'en-GB,en;q=0.9,en-US;q=0.8')
Cookie: ajs_anonymous_id=63d76b69-c268-4b63-aeeb-704fd15e24bc
(b'Cookie', b'ajs_anonymous_id=63d76b69-c268-4b63-aeeb-704fd15e24bc')
 in function ACCEPT_CONN
b'GET /assets/app_1024.png HTTP/1.1\r\n'Host: 192.168.1.206:8266
(b'Host', b'192.168.1.206:8266')
Connection: keep-alive
(b'Connection', b'keep-alive')
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/127.0.0.0 Safari/537.36 Edg/127.0.0.0
(b'User-Agent', b'Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/127.0.0.0 Safari/537.36 Edg/127.0.0.0')
Accept: image/avif,image/webp,image/apng,image/svg+xml,image/*,*/*;q=0.8
(b'Accept', b'image/avif,image/webp,image/apng,image/svg+xml,image/*,*/*;q=0.8')
Referer: http://192.168.1.206:8266/
(b'Referer', b'http://192.168.1.206:8266/')
Accept-Encoding: gzip, deflate
(b'Accept-Encoding', b'gzip, deflate')
Accept-Language: en-GB,en;q=0.9,en-US;q=0.8
(b'Accept-Language', b'en-GB,en;q=0.9,en-US;q=0.8')
Cookie: ajs_anonymous_id=63d76b69-c268-4b63-aeeb-704fd15e24bc
(b'Cookie', b'ajs_anonymous_id=63d76b69-c268-4b63-aeeb-704fd15e24bc')
 in function ACCEPT_CONN
b'GET / HTTP/1.1\r\n'Host: 192.168.1.206:8266
(b'Host', b'192.168.1.206:8266')
Connection: Upgrade
(b'Connection', b'Upgrade')
Pragma: no-cache
(b'Pragma', b'no-cache')
Cache-Control: no-cache
(b'Cache-Control', b'no-cache')
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/127.0.0.0 Safari/537.36 Edg/127.0.0.0
(b'User-Agent', b'Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/127.0.0.0 Safari/537.36 Edg/127.0.0.0')
Upgrade: websocket
(b'Upgrade', b'websocket')
Origin: http://192.168.1.206:8266
(b'Origin', b'http://192.168.1.206:8266')
Sec-WebSocket-Version: 13
(b'Sec-WebSocket-Version', b'13')
Accept-Encoding: gzip, deflate
(b'Accept-Encoding', b'gzip, deflate')
Accept-Language: en-GB,en;q=0.9,en-US;q=0.8
(b'Accept-Language', b'en-GB,en;q=0.9,en-US;q=0.8')
Cookie: ajs_anonymous_id=63d76b69-c268-4b63-aeeb-704fd15e24bc
(b'Cookie', b'ajs_anonymous_id=63d76b69-c268-4b63-aeeb-704fd15e24bc')
Sec-WebSocket-Key: mfaZXL12D3lbdFrabiVqew==
(b'Sec-WebSocket-Key', b'mfaZXL12D3lbdFrabiVqew==')
Sec-WebSocket-Extensions: permessage-deflate; client_max_window_bits
(b'Sec-WebSocket-Extensions', b'permessage-deflate; client_max_window_bits')
Sec-WebSocket-Key: b'mfaZXL12D3lbdFrabiVqew==' 24
respkey: b'jRKgcd09qKRrvj7KyBjWDZyx9k4='

WebREPL connection from: ('192.168.1.170', 56494)
dupterm: EOF received, deactivating

```

I have tried building my own version of Viper-IDE and extended the timeout on the readUntil() that is looking for Password (in transports.js) to a much longer period but the Timeout still occurs.

(Thanks for your efforts on Viper-IDE.)

vshymanskyy, 2024-07-31:
Thanks for additional inputs

ThinkTransit, 2024-07-31:
> Did you try the latest stable MicroPython build?

Yes I have tested with the latest stable version (v1.23.0) but the result is the same.

If there is any testing you would like me to do just let me know.

rimelis, 2025-01-28:
Hello,

I am also experiencing the same error on ESP-WROOM-32 having 1.24.1 MicroPython. I tested on Linux (Kubuntu) and Android phone, results are the same.

ppamment, 2025-05-16:
Same issue here on esp32-s3 devkit-C