vshymanskyy/ViperIDE #19: GitHub Repo Contents API

brainstorm, 2024-08-04:
Following up from https://github.com/vshymanskyy/ViperIDE/issues/17#issuecomment-226719210

brainstorm, 2024-08-04:
> The idea is cool 😎
But, i'm wondering how the sync could work. In instance, i have a GitHub repo with some files. Now, i'm connecting a device that may have a slightly (or completely) different file system. Which one is correct? Is this device even associated with the repository?
Same applies to regular host filesystem access, if/when it gets implemented.

Agree that it's tricky to implement without either loading complexity on either configuration or opinionated default conventions.

As a first draft on this idea I'd probably limit it to making sure that the board is blank and then assume that the GH repo is the authoritative [source of truth (SSOT)](https://en.wikipedia.org/wiki/Single_source_of_truth). Then provide "escape hatches" for those users that already have files on the board but want to use that feature at their own risk... although moving those files to a repo first and wiping the device is not too onerous anyway?

vshymanskyy, 2024-08-04:
Yes, it gets even more complicated if you have multiple boards running the same firmware.
It could easily lead to multiple boards running different (but clean) revisions of the same repo.

😵‍💫

brainstorm, 2024-08-04:
> Yes, it gets even more complicated if you have multiple boards running the same firmware. It could easily lead to multiple boards running different (but clean) revisions of the same repo.
> 
> 😵‍💫

In this case, wouldn't it be enough to save the HEAD hash to disambiguate state? Or worst case, keeping the `.git` folder around that can later be removed when shipping to production?

vshymanskyy, 2024-08-04:
No, keeping .git on the device is not gonna work