vshymanskyy/ViperIDE #23: Improve File Manager

vshymanskyy, 2024-08-11:
- [ ] Ensure that editor modifications are not lost when navigating or reconnecting the board
  - TBD: Warn about unsaved changes?
- [ ] Upload local files or folders
  - [ ] TBD: Drag'n'drop files into editor and file manager?
- [ ] Download FS snapshot as as zip/tar file
- More file operations
  - [ ] Rename file
  - [ ] Copy file

DeeJay, 2024-08-13:
" Download FS snapshot as as zip/tar file "

The **rshell** utility [https://github.com/dhylands/rshell](https://github.com/dhylands/rshell) has an **rsync** sub-command [https://github.com/dhylands/rshell#rsync](https://github.com/dhylands/rshell#rsync) which 
> Recursively synchronises a source directory to a destination.

This might either be a part of your solution, or inspection of the codebase might give you ideas for your own implementation?

I have been using rshell/rsync to keep a copy of the FS on an RPi Pico running MP 1.23 synchronised for backup on a host/workstation with a USB connection between the two devices. No archiving or compression involved.

sfe-SparkFro, 2025-08-25:
Adding additional file management requests:

* Deleting a folder should automatically and recursively delete all files and folders within it.
* Add option to create a new folder, not just new files.
* Cut files and folders, in addition to copy.
* Drag and drop files and folders to move them around.
* Collapsible folders (my board has dozens of files on it).

Boss-1s, 2025-11-18:
How about exporting and importing from the local machine?