vshymanskyy/ViperIDE #30: Errors when attempting connect to board running aiorepl

Josverl, 2024-09-09:
**Describe the bug**

**Setup**
 - MicroPython board: rp2040 
 - MicroPython version: 1.23.0
 - OS version: Windows
 - Browser: Edge
 - Connection type: USB
 - device running asyncio & aiorepl  on startup ( from main.py) , that changes the prompt to `--> `  (sys.ps1) 
   https://github.com/micropython/micropython-lib/tree/master/micropython/aiorepl

**To Reproduce**
Steps to reproduce the behavior:

2. Click on 'connect USB`
3. Connect USB alternates between green/yellow icons 
4. after a minute or so , sometimes I get connection with the repl , other times I only get connection errors 
5. at that time multiple `machine.reset()` have been issued 

**Expected behavior**
* to connect to the repl as provided by aiorepl and the prompt `--> ` be recognized as one of the standard prompts similar to `>>> ` 

**Actual Screenshots / Logs**
<details><summary>Terminal transcript</summary>
<p>

--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> Bleep , state: 0

--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> Bleep , state: 0

--> 
--> 
--> 
--> import machine; machine.reset()

--> 
--> 
--> 
--> Bleep , state: 0

--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> Bleep , state: 0

--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
--> 
MicroPython v1.23.0-1.g82f2d8332 on 2024-09-07; RP2040-Zero with RP2040
Type "help()" for more information.
>>> 
MPY: soft reboot
Setting color to (76, 76, 76)
Setting color to (0, 0, 0)
Starting tasks...
Bleep , state: 0
Starting asyncio REPL...
--> 
MicroPython v1.23.0-1.g82f2d8332 on 2024-09-07; RP2040-Zero with RP2040
Type "help()" for more information.
>>> 
>>> 
>>> 
MPY: soft reboot
Setting color to (76, 76, 76)
Setting color to (0, 0, 0)
Starting tasks...
Bleep , state: 0
Starting asyncio REPL...
--> 
MicroPython v1.23.0-1.g82f2d8332 on 2024-09-07; RP2040-Zero with RP2040
Type "help()" for more information.
>>> 
>>> 
>>> 
MPY: soft reboot
Setting color to (76, 76, 76)
Setting color to (0, 0, 0)
Starting tasks...
Bleep , state: 0
Starting asyncio REPL...
--> 
MicroPython v1.23.0-1.g82f2d8332 on 2024-09-07; RP2040-Zero with RP2040
Type "help()" for more information.
>>> 
>>> 
>>> 
MPY: soft reboot
Setting color to (76, 76, 76)
Setting color to (0, 0, 0)
Starting tasks...
Bleep , state: 0
Starting asyncio REPL...
--> 
MicroPython v1.23.0-1.g82f2d8332 on 2024-09-07; RP2040-Zero with RP2040
Type "help()" for more information.
>>> 
>>> 
>>> 
MPY: soft reboot
Setting color to (76, 76, 76)
Setting color to (0, 0, 0)
Starting tasks...
Bleep , state: 0
Starting asyncio REPL...
--> 
MicroPython v1.23.0-1.g82f2d8332 on 2024-09-07; RP2040-Zero with RP2040
Type "help()" for more information.
>>> 
>>> 
>>> 
MPY: soft reboot
Setting color to (76, 76, 76)
Setting color to (0, 0, 0)
Starting tasks...
Bleep , state: 0
Starting asyncio REPL...
--> 
MicroPython v1.23.0-1.g82f2d8332 on 2024-09-07; RP2040-Zero with RP2040
Type "help()" for more information.
>>> 
>>> 
>>> 
MPY: soft reboot
Setting color to (76, 76, 76)
Setting color to (0, 0, 0)
Starting tasks...
Bleep , state: 0
Starting asyncio REPL...
--> 
MicroPython v1.23.0-1.g82f2d8332 on 2024-09-07; RP2040-Zero with RP2040
Type "help()" for more information.
>>> 
>>> 
>>> 
MPY: soft reboot
Setting color to (76, 76, 76)
Setting color to (0, 0, 0)
Starting tasks...
Bleep , state: 0
Starting asyncio REPL...
--> 
MicroPython v1.23.0-1.g82f2d8332 on 2024-09-07; RP2040-Zero with RP2040
Type "help()" for more information.
>>> 
>>> 
>>> 
MPY: soft reboot
Setting color to (76, 76, 76)
Setting color to (0, 0, 0)
Starting tasks...
Bleep , state: 0
Starting asyncio REPL...
--> 
MicroPython v1.23.0-1.g82f2d8332 on 2024-09-07; RP2040-Zero with RP2040
Type "help()" for more information.
>>> 
>>> 
>>> 
>>> 
</p>
</details> 

**Additional context**
I assume that viper-ise waits for detection of a system prompt , and that the expected value is limited to ``>>> ` and possibly `... `.
aiorepl uses a [default prompt of `--> `](https://github.com/micropython/micropython-lib/blob/27e4d73bc2618d378a0610960cf5e81985e5d914/micropython/aiorepl/aiorepl.py#L96C24-L96C37)
In addition the micropython repl allows [the prompts to be changed](https://docs.micropython.org/en/latest/library/sys.html#sys.ps1) , so it would make sense to allow for prompts other than just `>>> `

vshymanskyy, 2024-09-10:
Aiorepl is not supported.
I dont think its supported by mpremote either

Josverl, 2024-09-11:
With mpremote it just works, (Jimmo worked on both aiorepl and mpremote)
Both the repl and copy work while keeping an async app running.

vshymanskyy, 2024-09-11:
Interesting, thanks for bringing it up!