vshymanskyy/ViperIDE #32: ws:// urls don't work for web-repl bridge in chrome as the viper-ide site always redirects to https and therefore requires wss: links not ws:

tyeth, 2024-09-12:
Just discovered the Bridge mode / page at https://viper-ide.org/bridge.html and thought I'd attempt to connect to the serial port / repl of a circuitpython device, as it runs the web workflow and therefore has a websocket open at ws://192.168.50.160/cp/serial/

It fails due to "Connection to an unsecure WebSocket is blocked on a secure website".

I guess it could be made to work, but probably offers little benefit at the moment.

Thanks for the great tool!

vshymanskyy, 2024-09-12:
Yes, this is a conceptual limitation, and it's documented.
With some efforts, you can run ViperIDE locally in an insecure context, and it should work with such devices

tyeth, 2024-09-12:
Thanks for your reply!
Yeah that makes sense, and nice one documenting it (can't deny I haven't
read all the docs yet).
I think I may revisit this once I want to run it from/on device, or have a
more pressing need for Circuitpython usage over the network

vshymanskyy, 2024-09-12:
There's not much sense in running it on the device, it's ~10MB. Unless you want to sacrifice almost all of the goodness...
Well, it could be shrinked to ~100Kb (as it used to be in the early days).

tyeth, 2024-09-12:
Yikes, not small then! Only fit on a couple of my larger esp devices unless
I go for the lofi version.

closed 2024-09-12 by tyeth