vshymanskyy/ViperIDE #33: A few observations and suggestions.

robert-hh, 2024-09-16:
Since my first tests ViperIDE has improved well. For me it looks much better than e.g. Thonny, especially because it has no hidden actions under the hood like Thonny does. A few observations and suggestion:

- If a file from the board has been displayed and the deleted, the file content is still visible in the Edit window. That does not seem consistent.
- If a device is disconnected, the files of the file pane are still visible. The file pane window should be cleared.
- A method for manually refreshing the file pane. Could be for instance a click on the file folder symbol at the lower left of the window. If file names or files change due actions at the REPL promt or running code, this does not visible until the board is disconnected and connected again.
- Copy files between the board and the PC's file system. May be a problem with mobile devices. Could be done by drag and drop between an file explorer window and the app's file pane instead of a separate dialogue.

vshymanskyy, 2024-09-17:
@robert-hh thanks for your feedback, I think it's very inline with what I'm thinking.
1. I have no clear picture of how this could be implemented efficiently, without re-scanning the whole FS every time a user script is run (and basically, every terminal command).
2. Needs to be fixed, I think i have a good idea on how to do it, but it requires quite a lot of stuff to happen
3. If we add a manual refresh button, will it fix point n.1 ?
4. Already discussed in #12, #23

robert-hh, 2024-09-17:
ad 1: I was just thinking of a manual delete in the ViderIDE file pane. Then you have the context. The file list is already updated. Just the edit window has to be cleared. That is slightly different from topic 3, where changes to the file system is cause by code running on the board.
Ad 3: See above. I do not know if  a refresh button will solve topic 1.

vshymanskyy, 2024-09-17:
In general, I think of reworking the board connection mechanism:
1. Upon connection verify if user code is running (tap `Enter`, see if REPL responds)
   - If REPL responds, automatically load system info and FS structure
2. If no REPL responce, check if user have selected the `Interrupt Running Code` setting
   - If no interruption is needed, stay in Terminal mode, disable FS and Editor temporarily until the code exits
3. If interrupting is needed, interrupt code, load FS contents

Also, I think `auto-restart` setting would be useful, that would soft-reset board when:
- initial connection interrupted the app execution
- user saves any file

In general, I'm thinking about adding Simple/Advanced mode setting that would optimize the UX for a specific use case

vshymanskyy, 2024-09-17:
> manual delete in the ViderIDE file pane. Then you have the context.

Yes, I got it now. This will be solved as part of https://github.com/vshymanskyy/ViperIDE/issues/28

vshymanskyy, 2024-09-17:
So from above comments, I think all items are covered by other tickets, except
- A method for manually refreshing the file pane

This is very easy to add, I'll do it right away

vshymanskyy, 2024-09-17:
[image]

@robert-hh fyi

closed 2024-09-17 by vshymanskyy

robert-hh, 2024-09-17:
Thanks. Looks good. Don't forget to update ViperIDE.org

vshymanskyy, 2024-09-17:
All changes go directly to ViperIDE.org 🤪

robert-hh, 2024-09-17:
I tried it and it was not present. But maybe I forgot to refresh the Browser so see the refresh button.