vshymanskyy/ViperIDE #42: ✨ Allow project import from GitHub

vshymanskyy, 2024-10-14:
Once #41 is ready, it should be rather easy to implement project import:
- [ ] Define a simple `project.json` structure. This can be inspired/adapted from mip, node.js, PlatformIO etc.
- [ ] Implement the project import mechanism. This is very similar to package installation
  - [ ] Install project dependencies (libraries)
- [ ] Improve UI/UX
  - [ ] Display `README.md` from the sample
  - [ ] Connect board step
  - [ ] Install projects
- [ ] Project sharing via link, i.e. `https://viper-ide.org/?import=https://github.com/emlearn/emlearn-micropython/tree/master/examples/soundlevel_iir`

vshymanskyy, 2024-10-14:
`package.json` suggestion:

```json
{
    "name": "MySample",
    "version": "master",
    "description": "Makes everyone happy",
    "author": "Citizen of the Universe",
    "license": "MIT",
    "readme": "",
    "files": [
        "main.py", "cfg/config.json", "www/logo.png"
    ],
    "urls": [
      ["some_file.py", "github:org/repo/dir/file.py"],
      ["data.dat", "github:org/repo/dir/file.dat"]
    ],
    "deps": [
    ]
}
```

Notes:
- `files` is a simpler/shorter version of `urls`
- `urls`, `deps` are the same as described in https://docs.micropython.org/en/latest/reference/packages.html
- When ViperIDE (and later `mpremote`) installs a project (not a package), target installation path should `/` or `/flash` by default

Comments and ideas are welcome

vshymanskyy, 2024-10-14:
@RobertBlackhart I think this would be very useful for education scenarios

RobertBlackhart, 2024-10-14:
I agree. I really like the ability to link using a URL query. I'd be happy to do some work on #41/#42 if you're looking for volunteers, although I can't promise a timeline right now (not that you're asking, just wanted to set expectations).

vshymanskyy, 2024-10-14:
No, this is completely on me for now. Also, it's in discussion/definition phase (also setting expectations here :) )

vshymanskyy, 2024-10-18:
There's now an experimental support of installing applications.
For example you can install [`aiodns` example](https://github.com/vshymanskyy/aiodns/tree/main/examples/simple) using this button:

[[image]](https://viper-ide.org/?install=github:vshymanskyy/aiodns/examples/simple)

jonnor, 2024-11-03:
I am adding Run in ViperIDE to another example in emlearn-micropython. Ref this package.json, https://github.com/emlearn/emlearn-micropython/blob/master/examples/har_trees/package.json
However, this example one depends on a couple of libraries which are not in micropython-lib, and fails to install because the dependencies are not known.
The libraries in question are:
- https://github.com/jonnor/micropython-zipfile
- https://github.com/jonnor/micropython-npyfile
And I also plan to utilize these also soon:
- https://github.com/jonnor/micropython-mpu6886
- https://github.com/jonnor/bma423-pure-mp

How to add these?

jonnor, 2024-12-03:
> How to add these?

Using the URL works! For example `"deps": [ [ "https://github.com/jonnor/micropython-mpu6886", "master" ] ]`. Don't know why I did not try it before...