vshymanskyy/ViperIDE #44: Changes to imported files not reflected in running code

bitbronze, 2024-10-17:
Changes to imported files are not reflected in running code

**Setup**
 - MicroPython board: ESP32-C6
 - MicroPython version: latest/preview
 - OS version: Windows
 - Browser: Chrome
 - Connection type: USB

**To Reproduce**
Steps to reproduce the behavior:
1. Create file "importtest.py" with the following code:
`def somefunc():`
`    print("123")`

2. Create file "importtestrun.py" with the following code:
`import importtest`
`importtest.somefunc()`

3. Run "importtestrun.py", and see that the behavior is as expected
4. Change "importtest.py"
5. Run "importtestrun.py", and see that the behavior is incorrect. (its still running old code)

**Expected behavior**
Changes in importtest.py should be reflected when importtestrun.py is run. Unfortunately this is not the case and the device needs to be disconnected and reconnected to resolve the issue and the import to work as expected with the new changes

DeeJay, 2024-10-17:
_{Disclaimer: I am a barely competent beginner with Python/MicroPython.}_

Might that be the actual 'expected' behaviour? (And that **your** expectation is incorrect?)

With full-fat Python I think **import**  is a 'lazy' operation that will not be repeated if the corresponding namespace already exists?

Perhaps micropython is using the same behaviour?

ma261065, 2024-10-18:
You should just be able to do a soft reset on the device instead of dis/re-connecting.

As far as I understand it, this is by design.

vshymanskyy, 2024-10-18:
Yes, this is how MicroPython itself works.
You need to soft-reset ant this is why you have a dedicated button in the upper right corner of the Terminal (and the Ctrl+D shortcut)

ma261065, 2024-10-18:
Well you can see the same behavior in just the REPL...

Put a `print('Hello')` at the start of `importtest.py`, then at a REPL prompt type `import importtest` a few times.

You will see that it only prints `Hello` once

vshymanskyy, 2024-10-18:
Adding an option to auto-soft-reset is easy