vshymanskyy/ViperIDE #48: [Enhancement] "Save" Indicator

ma261065, 2024-11-04:
When saving small files, the "Save Completed" popup occurs pretty well straight away. So, no problem...

However, for large files the saving process can take many tens of seconds and it's not obvious that anything is happening. So a novice user is probably going to think their save hasn't worked and click the Save Button again, and again....

It would be useful to have a spinner, or preferably a progress bar of some sort, either in the file tree or as a whole-of-app overlay to show that a save is in progress.

vshymanskyy, 2025-01-31:
Right now, the activity is indicated by changing the color of the Connectivity (USB, BLE, Link) icon.
It changes from green to orange when the device is busy.
I agree that having a progress bar would be great, but I'm not sure when I'll be able to get to it.

ma261065, 2025-01-31:
Ahh, I hadn't even noticed that. Probably due to my red/green colourblindness...

Thanks for pointing it out.

vshymanskyy, 2025-01-31:
Which colors would you suggest to replace the `green/yellow` combination with?

ma261065, 2025-01-31:
Personally, I find blue to be a good colour that is very different to either red or green.

vshymanskyy, 2025-01-31:
- `#1aafff` - connected, passive
- `#ffa500` - connected, active