vshymanskyy/ViperIDE #50: ViperIDE implementation of WebREPL consistently fails with ESP32-C3

jeffstearns, 2025-04-13:
**Describe the bug**
The ViperIDE implementation of WebREPL consistently fails with an ESP32-C3.
Shortly after connecting to the board, the browser will post an error message saying "Error: Timeout reached before finding the ending sequence".
This happens consistently for every connection attempt.

**Setup**
 - OS version: MacOS Sequoia 15.3.2 on MacBook Pro 16" with 64GB RAM
 - Browser: Tested with Chrome version 134.0.6998.166 (Official Build) (x86_64), Safari Version 18.3.1 (20620.2.4.11.6), and Firefox 137.0 (64-bit)
 - MicroPython board: ESP32-C3 from seeed studio
 - Connection type: WebREPL

**To Reproduce**
Steps to reproduce the behavior:
1. Obtain an ESP32-C3 from seeed studio or similar.
2. Connect via USB.
3. Clear the flash to remove any possible residual cruft.
4. Install the current MicroPython version 1.24.1 for ESP32-C3 using Thonny or equivalent installer.
5. Follow the instructions at https://github.com/vshymanskyy/ViperIDE/blob/main/docs/Web-REPL-Server.md

**Expected behavior**
ViperIDE connects to the board via WebREPL and displays a functional REPL window.

**Additional context**
I have repeated this test several dozen times, testing power cycling, soft resets, hard resets, firmware erasures, and viper-tools package reinstalls.  None of these efforts have ever produced a working ViperIDE WebREPL connection.

ViperIDE has no trouble connecting to the board via USB.  USB connections always work.  It's specifically WebREPL connections that don't work.

TCP connectivity between the host computer and the ESP32-C3 is good.  I have confirmed this with ping/telnet/Python.  There's no problem establishing and maintaining a TCP connection to the ESP32-C3.  It's the WebREPL protocol itself that's having trouble.

_**Please let me know what additional information or tests would be helpful.  Thank you!**_

**Actual Screenshots / Logs**

Here's the javascript console from a Chrome browser.  Console errors from other browsers are similar.  The error always points to a "Timeout reached before finding the ending sequence".
```
built glob set; 25 literals, 0 basenames, 0 extensions, 0 prefixes, 0 suffixes, 0 required extensions, 0 regexes
(index):4 built glob set; 0 literals, 1 basenames, 3 extensions, 0 prefixes, 0 suffixes, 0 required extensions, 0 regexes
22(index):2 Error: Timeout reached before finding the ending sequence
    at Vx.readUntil ((index):4:124229)
    at async Yx.interruptProgram ((index):4:131844)
    at async Yx.enterRawRepl ((index):4:132125)
    at async Yx.begin ((index):4:131632)
    at async I$ ((index):10:73253)
    at async (index):10:90650 'Error: Timeout reached before finding the ending sequence\n    at Vx.readUntil (http://10.5.4.95:8266/:4:124229)\n    at async Yx.interruptProgram (http://10.5.4.95:8266/:4:131844)\n    at async Yx.enterRawRepl (http://10.5.4.95:8266/:4:132125)\n    at async Yx.begin (http://10.5.4.95:8266/:4:131632)\n    at async I$ (http://10.5.4.95:8266/:10:73253)\n    at async http://10.5.4.95:8266/:10:90650'
O_ @ (index):2

18(index):2 Error: Timeout reached before finding the ending sequence
    at Vx.readUntil ((index):4:124229)
    at async Yx.interruptProgram ((index):4:131844)
    at async Yx.enterRawRepl ((index):4:132125)
    at async Yx.begin ((index):4:131632)
    at async I$ ((index):10:73253)
    at async (index):10:90650 'Error: Timeout reached before finding the ending sequence\n    at Vx.readUntil (http://10.5.4.95:8266/:4:124229)\n    at async Yx.interruptProgram (http://10.5.4.95:8266/:4:131844)\n    at async Yx.enterRawRepl (http://10.5.4.95:8266/:4:132125)\n    at async Yx.begin (http://10.5.4.95:8266/:4:131632)\n    at async I$ (http://10.5.4.95:8266/:10:73253)\n    at async http://10.5.4.95:8266/:10:90650'
O_ @ (index):2
interruptProgram @ (index):4

(index):2 Error: Error: Board is not responding
    at (index):2:1122599 'Error: Error: Board is not responding\n    at http://10.5.4.95:8266/:2:1122599'
O_ @ (index):2
(anonymous) @ (index):2
```

MirrorCY, 2025-06-24:
same issue

d51r3verse, 2025-12-24:
Same here too.
Chrome 143.0.7499.170
Site settings: allow all
```
Access to fetch at 'https://freeipapi.com/api/json' from origin 'http://192.168.111.111:8266' has been blocked by CORS policy: No 'Access-Control-Allow-Origin' header is present on the requested resource.
...
19(index):2 Error: Timeout reached before finding the ending sequence
    at Vx.readUntil ((index):4:124229)
    at async Yx.interruptProgram ((index):4:131844)
    at async Yx.enterRawRepl ((index):4:132125)
    at async Yx.begin ((index):4:131632)
    at async I$ ((index):10:73253)
    at async (index):10:90650 'Error: Timeout reached before finding the ending sequence\n    at Vx.readUntil (http://192.168.111.111:8266/:4:124229)\n    at async Yx.interruptProgram (http://192.168.111.111:8266/:4:131844)\n    at async Yx.enterRawRepl (http://192.168.111.111:8266/:4:132125)\n    at async Yx.begin (http://192.168.111.111:8266/:4:131632)\n    at async I$ (http://192.168.111.111:8266/:10:73253)\n    at async http://192.168.111.111:8266/:10:90650'
```
Same as Chrome alternative other browsers and mobile latest Chrome too.

Josverl, 2026-02-14:
> * Root cause: machine.Timer(-1) not available on latest micropython. that's why WSS module load fails.
>   Solution: -1 to 0

That is not quite the case. Actually `machine.Timer(-1)` is not supported on all ports / boards

From: https://docs.micropython.org/en/latest/library/machine.Timer.html#machine.Timer
> On ports which support virtual timers the id parameter is optional - the default value is -1 which constructs a virtual timer.
> On ports which support hardware timers, setting the id parameter to a non-negative integer determines which timer to use.

So likely the code would need to improved to something like:
```py
for _id in [-1]+list(range(16,0,-1)):
    try:
        timer_ = machine.Timer(id)
        break
    except ValueError:
        pass
```